MarlinFirmware/U8glib-HAL #8: [BUG] [dev] STM32F1 and SSD1306 compile problem

malloc-19, 2019-04-18:
### Description

It looks like #6 bricked my ability to us my Blue Pill (STM32F103CB) with an SSD1306 OLED.
(https://github.com/MarlinFirmware/U8glib-HAL/commit/28e15cd1635354c90f266777c8f9a144717622b2)

By default "U8glib-HAL" is being ignored when compiling for STM32F1 but I found that it did work anyway before the Changes in #6 were made.

### Steps to Reproduce

1. Get new Marlin bugfix-2.0.x code and open in VS Code
2. Make the following changes in "platformio.ini"
 - under [platformio]
   - env_default = STM32F1

 - under: [env:STM32F1] 
   - change "lib_ignore  =  U8glib-HAL" to "lib_ignore  =  ;U8glib-HAL"
   - change "board  =  genericSTM32F103RE" to "board  =  genericSTM32F103CB"

3. Make the following changes in "Configuration.h"
 - "#define SERIAL_PORT 0" to "#define SERIAL_PORT 1"
 - "#define MOTHERBOARD BOARD_RAMPS_14_EFB" to "#define MOTHERBOARD BOARD_MORPHEUS"
 - "//#define U8GLIB_SSD1306" to "#define U8GLIB_SSD1306"

**Expected behavior:**

I expected the Code to Compile.

**Actual behavior:**

It didn't :)
Some Errors from the Console:

C:\users\[NAME]\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/wirish_types.h:65:9: error: unknown type name 'bool'

C:\users\[NAME]\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'

C:\users\[NAME]\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/wirish_math.h:50:6: error: conflicting types for 'random'

*** [.pioenvs\STM32F1\libe06\U8glib-HAL_ID1932\clib\u8g_com_arduino_common.c.o] Error 1
*** [.pioenvs\STM32F1\libe06\U8glib-HAL_ID1932\clib\u8g_com_arduino_fast_parallel.c.o] Error 1

Full console:
[ERROR_STM32F1_SSD1306.txt](https://github.com/MarlinFirmware/U8glib-HAL/files/3095301/ERROR_STM32F1_SSD1306.txt)

####HOTFIX

Since an older Version did compile, I was able to find where the error comes from. Here is how to get it working:

1. Coplete: "### Steps to Reproduce"
2. Get U8glib-HAL code from: November 2018: https://github.com/MarlinFirmware/U8glib-HAL/tree/99e44fa4903174ce8660860ee166504e13d0483a
3. Replace "src" and "library.json" in "...\Marlin-bugfix-2.0.x\.piolibdeps\U8glib-HAL_ID1932" with files from Nov 2018
4. Restart VS Code

#### Additional Information

I am not sure why #6 causes these weird errors.

MuellerDominik, 2019-04-26:
@malloc-19, I can see that you're using an old version of the `ST STM32` platform (`arduinoststm32@2.1.180219`). I strongly recommend using the latest version of the `ST STM32` platform: `v5.3.0`.

I followed your "Steps to Reproduce" with the only difference being that I changed the following line in the `platformio.ini` file under the `STM32F1` environment:
* `platform = ststm32@<4.4.0` to `platform = ststm32@5.3.0`

This worked for me with no build errors. Unfortunately, I couldn't verify that it actually works because I don't have a Blue Pill and neither do I have a SSD1306 OLED.

Don't forget to make sure that you're using the latest version of the `U8glib-HAL` extension from the `dev` branch (with https://github.com/MarlinFirmware/U8glib-HAL/pull/6 merged). You can do this by deleting `.piolibdeps\U8glib-HAL_ID1932` as discibed here: https://github.com/MarlinFirmware/Marlin/commit/1de6e7fb9d7a4eef4f10125a05ec608aca9d5168#commitcomment-33192679.

malloc-19, 2019-05-02:
I can confirm that 5.3.0 works. Thanks!

Is there a particular reason why it's <4.4.0 by default?

closed 2019-05-11 by malloc-19