MarlinFirmware/U8glib-HAL #9: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER Random Movement

SISLANGER, 2019-04-27:
Hi, so the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is working fine, except the menu is erratic, sometime it will move to next item, sometimes it won't, sometimes it will jump four or five spaces after not doing anything.

Push click ok, display ok, just the scrolling issue

Bob-the-Kuhn, 2019-04-27:
Have you tried playing with the settings in the **ENCODER SETTINGS** in **configuration.h**?

SISLANGER, 2019-04-27:
Yes up and down with the settings, but still no joy sometimes will go the wrong way was fine on the gen 1.4

4 pulses is the best yet but still 50/50  it will go where I want it to go.

Bob-the-Kuhn, 2019-04-28:
The BTN_EN1 & BTN_EN2 signals are just switch contacts.  Perhaps the problem is the pull up resistors within the CPU are too weak to overcome the noise.  I suggest adding 1K - 10K pullup resistors to those lines.

On way to access the signals is to press another 10 pin connector onto the EXP2 cable and then create some insulated resistor jumpers to go from +5V to pins 3 and 5.

SISLANGER, 2019-04-28:
@bob-the-Kuhn thank you will give that a go.

SISLANGER, 2019-05-09:
Turns out it was the LCD got a replacement and not a problem, strange it was working fine on the Gen 1.4 an hour before hand.

closed 2019-05-09 by SISLANGER