MarlinFirmware/U8glib-HAL #10: [BUG] Warnings when build

sl1pkn07, 2019-05-26:
~~~
Compiling .pioenvs/LPC1768/libe04/U8glib-HAL_ID1932/clib/u8g_com_msp430_hw_spi.c.o
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
 static uint8_t u8g_i2c_opt;   /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */
                ^~~~~~~~~~~
~~~

~~~
Compiling .pioenvs/LPC1768/libe04/U8glib-HAL_ID1932/clib/u8g_dev_ssd1309_128x64.c.o
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:153:22: warning: 'u8g_dev_ssd1306_128x32_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_univision_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:83:22: warning: 'u8g_dev_ssd1306_128x32_adafruit2_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit2_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:48:22: warning: 'u8g_dev_ssd1306_128x32_adafruit1_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit1_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/LPC1768/libe04/U8glib-HAL_ID1932/clib/u8g_dev_ssd1322_nhd31oled_bw.c.o
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x64.c:146:22: warning: 'u8g_dev_ssd1306_128x64_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x64_univision_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x64.c:44:22: warning: 'u8g_dev_ssd1306_128x64_adafruit1_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x64_adafruit1_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

~~~
Compiling .pioenvs/LPC1768/libe04/U8glib-HAL_ID1932/clib/u8g_u16toa.c.o
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_rot.c:48:27: warning: initialization of 'uint8_t (*)(u8g_t *, u8g_dev_t *, uint8_t,  void *)' {aka 'unsigned char (*)(struct _u8g_t *, struct _u8g_dev_t *, unsigned char,  void *)'} from incompatible pointer type 'uint8_t (*)(void *, void *, uint8_t,  void *)' {aka 'unsigned char (*)(void *, void *, unsigned char,  void *)'} [-Wincompatible-pointer-types]
 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };
                           ^~~~~~~~~~~~~~~~~~~~
.piolibdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_rot.c:48:27: note: (near initialization for 'u8g_dev_rot.dev_fn')
~~~

Using stock CR10 LCD in ReARM+ramps1.4 (https://github.com/sl1pkn07/Marlin/commit/7441c1500e02ebffb35e3f3a080cf218633ba44e)

ETE-Design, 2019-08-24:
Same here... Error...

tpruvot, 2019-08-29:
made a PR for i2c ones https://github.com/MarlinFirmware/U8glib-HAL/pull/12

closed 2019-09-01 by thinkyhead

sl1pkn07, 2019-09-02:
the warnings still here (remove .pio folder and download again)

thinkyhead, 2019-09-04:
These warnings are not important. Please ignore.