MarlinFirmware/U8glib-HAL #19: https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip : error 404

yet-another-average-joe, 2020-06-21:
Hello,

It seems the U8glib-HAL library for Marlin bugfix disappeared.

Compiling Marlin (VSCode + PIO) gives this error :

`FDUnrecognizedStatusCode: Got an unrecognized status code '404' when downloaded https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip`

typing "https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip" in a browser also gives a 404 error (and by the way this link is a broken link)

As a result, U8glib-HAL is not installed.

Never had this issue before... Was a "archive" or "bugfix" branch recently removed from this repo ?

richbria, 2020-06-21:
I just got here due to the same problem when attempting to use Marlin Firmware Auto Build to build for a sanguino1284p.

richbria, 2020-06-21:
Looks like the bugfix branch was merged in here so it is probably actually an issue with Auto Build.  @yet-another-average-joe did you get here using Auto Build as well?

https://github.com/MarlinFirmware/U8glib-HAL/pull/18

richbria, 2020-06-21:
I am incorrect. This is a problem with platforio.ini in Marlin 2.0.  I'm working on a fork now.

closed 2020-06-21 by thinkyhead

yet-another-average-joe, 2020-06-21:
OK ! bugfix branch is back !
I was just attempting to compile while thinkyhead was updating the repo.