MarlinFirmware/U8glib-HAL #26: arduino ide and U8glib-HAL

ellensp, 2021-06-24:
Since this is now the required U8glib for Marlin, @thinkyhead can you please make a decision about it.

The options I see are:

a) Officially drop support for building Marlin on Arduino IDE  (preferred option by most of us)
Presently the Arduino IDE will try and use the standard U8glib, Which is no longer compatible with Marlin and errors causing much user strife, see issues.   

b) Add a library.properties file into this repository so Arduino IDE users can just drop this version of the library into their Arduino/libraries folder to provide Marlin with the correct U8glib required to compile.   

Example library.properties here https://github.com/MarlinFirmware/Marlin/issues/22218#:~:text=library.properties.zip

Related issues:
https://github.com/MarlinFirmware/Marlin/issues/22218
https://github.com/MarlinFirmware/Marlin/issues/21967
https://github.com/MarlinFirmware/Marlin/issues/21948
https://github.com/MarlinFirmware/Marlin/issues/21995

ellensp, 2021-07-09:
It seems from https://github.com/MarlinFirmware/Marlin/issues/22303#issuecomment-876884696 you have elected to continue fighting with Arduino-IDE

ellensp, 2021-07-10:
Has been added

closed 2021-07-10 by ellensp

Wallby, 2021-09-28:
NOTE: looks like this is not in [U8glib-HAL-0.4.5](https://github.com/MarlinFirmware/U8glib-HAL/releases/tag/0.4.5) or earlier. I copied the library.properties from [U8glib-HAL-0.5.0](https://github.com/MarlinFirmware/U8glib-HAL/releases/tag/0.5.0) for use with 0.4.5, which worked.