MarlinFirmware/U8glib-HAL #32: Typo in rc/clib/u8g.h#L451

ellensp, 2023-12-27:
u8g.h Has 

```
/* OLED 128x64 Display with SSD1309 Controller - Fast Single Page */
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_sw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_hw_spi;
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_sw_spi;
```
But last line should be
```
extern u8g_dev_t u8g_dev_ssd1309_128x64_f_i2c;
```

Also in platformio this changed u8g.h is not present in the downloaded library, 
.pio/libdeps/mks_robin_nano_v1v2/U8glib-HAL/src/clib/u8g.h
which I cannot explain, its there if I download the library manually

dbuezas, 2023-12-27:
Oh no, that's my fault. I don't have a computer at the moment to make a new PR, sorry for that.

thisiskeithb, 2023-12-27:
There may have been some force-pushing going on since https://github.com/MarlinFirmware/U8glib-HAL/commit/e258d50d318d75a36856d86767be4b9b52f50998 and https://github.com/MarlinFirmware/U8glib-HAL/commit/36f76705c570cedf01b315c5927871de324e30dc exist with a "Version 0.5.3" commit title. I suspect PIO isn't picking up the updated code and another tag is in order. cc: @thinkyhead

thinkyhead, 2023-12-27:
> Oh no, that's my fault.

_Mea culpa_. The buck stops here. I should have actually done some builds before merging.

closed 2023-12-27 by thinkyhead