MarlinFirmware/U8glib-HAL #33: U8glib-HAL  0.53 fails on Marlin STM builds

nicwilson58, 2023-12-28:
Visual Code informs this library was updated 22 hours ago from  0.52  to  0.53 but the libaray is broken for all STM compiles I try,  forcing a roll back to 0.52 resolves the issue:  

The error is below:

In file included from Marlin\src\lcd/dogm/u8g_fontutf8.h:11,
                 from Marlin\src\lcd/lcdprint.h:53,
                 from Marlin\src\lcd/marlinui.h:74,
                 from Marlin\src\MarlinCore.cpp:64:
.pio\libdeps\STM32F103RE_creality\U8glib-HAL\src/U8glib-HAL.h: In member function 'void U8GLIB_SSD1309_128X64_F::init(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
.pio\libdeps\STM32F103RE_creality\U8glib-HAL\src/U8glib-HAL.h:944:110: error: 'u8g_dev_ssd1309_128x64_f_sw_spi' was not declared in this scope; did you mean 'u8g_dev_ssd1309_128x64_sw_spi'?
  944 |   void init(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) { U8GLIB::init(&u8g_dev_ssd1309_128x64_f_sw_spi, sck, mosi, cs, a0, reset); }
      |                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                              u8g_dev_ssd1309_128x64_sw_spi

thisiskeithb, 2023-12-28:
Duplicate of https://github.com/MarlinFirmware/U8glib-HAL/issues/32

A new version of U8glib has already been tagged, but PIO is slow to pick it up / add it to the registry.

nicwilson58, 2023-12-28:
Just wondering,  does anyone have any idea how long a new tag usually takes to be picked up?  I would have expected a couple of hours,  but so far 0.54 is not seem by any of my PCs

nicwilson58, 2023-12-28:
Ignore my last,   the 0.54 build just got added,  and the issue is resolved.

thisiskeithb, 2023-12-28:
> Ignore my last, the 0.54 build just got added, and the issue is resolved.

Great! Since I don't have full access to this repo yet, please close the issue.

closed 2023-12-28 by nicwilson58